iNavFlight/blackbox-tools #12: GPS not logging valid timestamp

Safranek42, 2020-06-19:
Hi,

I tried blackbox logging and then decode the logged data to .csv files with blackbox tools.

However, the two output file for gps data lacks any meaningful timestamp. In the csv file, the 'time (us)' field is constant zero. In the gps file, <time>2000-01-01T00:00:00.000000Z</time> is the timestamp in every row. On the other csv file there is correct time data.

Does anyone know why is this happening, or how can it be solved?

Thanks in advance for any help.

stronnag, 2020-06-19:
I'll come up with a more sensible workaround for the log-rate == 1  issue; we don't need micro-second precision in the GPX, and I'll ignore @Safranek42 's issue if he declines to provide a log. Private email statements without data are not useful.

digitalentity, 2020-06-19:
Maybe we should just write a timestamp to every GPS frame. millis() since boot will be accurate enough. WDYT?

digitalentity, 2020-06-19:
This will break the blackbox-tools and mwp though

stronnag, 2020-06-19:
breaking mwp is not an issue, I do it all the time :)

AFAIK, we already have all the info in the log, A usec time stamp and a UTC start time. I think we can resolve this in BBL tools.

stronnag, 2020-06-19:
Unintended consequence of increasing MAX_SERVOS from 8 to 16 in the firmware and missing a knock-on effect in `blackbox_decode`.   

With this addressed, @Dope-johnny 's log generates a GPX with properly incrementing time stamps. 

PR to follow ... soon.

stronnag, 2020-06-19:
Static timestamps fixed by https://github.com/iNavFlight/blackbox-tools/pull/13

Marvellous, fixes the amazing bogus / backwards timestamps and duplicate iteration numbers shown in https://github.com/iNavFlight/blackbox-tools/issues/12#issuecomment-646806557 as well.

stronnag, 2020-06-19:
@Dope-johnny attached is a win64 binary (albeit built with msys2 rather than the MS compiler). 
Generates a correct GPX from your log in my win10 VM. Can you please check it?

[bbl-decode-win64.zip](https://github.com/iNavFlight/blackbox-tools/files/4806620/bbl-decode-win64.zip)

Dope-johnny, 2020-06-20:
@stronnag Okay, I got it running (eventually) and it works. Timestamps are correct and the .gpx file works fine in Google Earth. Also tried it with another log - all marvelous. 👌 Thanks a lot for fixing it so fast.

stronnag, 2020-06-20:
Thanks for the confirmation. As the only issue for which data has been provided has a fix merged, I'm closing the issue.

Should either of other respondents find that the fix does not address their problems and they're able to provide logs illustrating the problem, please open a new issue in this repository.

closed 2020-06-20 by stronnag