iNavFlight/blackbox-tools #26: presumable Incorrect parsing of GPS altitude

Torchi, 2020-12-28:
I noticed that GPS altitude in the GPX files produced by INAV blackbox decoder appear multiplied by 10. E.G. if altitude in the goggles was 100 meters, the GPX file reports 1000.    I played doing some tests crossing - Betaflight and INAV produced blackbox files - with - Betaflight and INAV blackbox decoders -.  All codes and files were relatively recent.   The test results have been:

decoder      blackbox         GPX 
for              produced by   altitude is
--------------------------------------
INAV           INAV              NOK     
INAV           Betaflight       NOK     
Betaflight     Betaflight      OK      
Betaflight     INAV             OK      

I did not debug code but I have seen in the Betaflight decoder source, commented  that from rel. 4 onward "GPS_altitude" is coded in the blackbox in decimeters, not centimeters.  The decoder for Betaflight does therefore  correctly the conversion in meters.  
The INAV decoder does no conversion. 
An hypothesis:  also INAV encodes now "GPS_altitude" in the blackbox in decimeters   (and this is the reason the actual decoder reports it x10)  ?

stronnag, 2020-12-28:
This is not the case, inav blackbox decoder reports the GPS altitude in m. The site logged in the image is 38m AMSL.
[image]
Please provide an example  inav log that supports your supposition in order that it can be investigated.

Torchi, 2020-12-28:
You are right !!!, sorry.  I changed one particular model from Betaflight  to INAV and when I recovered the Blackbox file from G-Drive I did not notice that the recording was dated some days earlier I changed the model into INAV.  Yes, in the log internal file header I have seen now that it is written "Firmware revision:  Betaflight 4.2.0 .." but earlier I overlooked that.  Just now I have armed that model, recovered the blackbox file and  altitude is correct. You can close the report,  for me.

stronnag, 2020-12-28:
No problem. Thanks for confirming.

closed 2020-12-28 by stronnag