iNavFlight/blackbox-tools #42: BB log doesn't decode but opens without problems in explorer

avsaase, 2021-08-14:
When decoding the attached log with `blackbox_decode` it prints
```
Log 1 of 1Statistics
I frames       1  112.0 bytes avg      112 bytes total
H frames       9    1.0 bytes avg        9 bytes total
G frames      13    1.0 bytes avg       13 bytes total
E frames       2    4.5 bytes avg        9 bytes total
S frames      10    1.0 bytes avg       10 bytes total
Frames         1  112.0 bytes avg      112 bytes total
Data rate: Unknown, no timing information available.

29011 frames failed to decode, rendering 31805 loop iterations unreadable.
```
and produces a CSV file that's only a few KB in size. However, when I open the log in INAV Blackbox Explorer I am not able to find any problems with the log (only a few dropped frames at 00:09.210). Betaflight Blackbox Explorer on the other hand is also not able to open the log. 

I am using a serial logger and I know I am pushing it with the 250Hz logging rate but I would think if I can view the log I would be able to decode it. Is this a problem with my log?

[LOG009.TXT](https://github.com/iNavFlight/blackbox-tools/files/6987256/LOG009.TXT)

stronnag, 2021-08-15:
Seems OK here.

```
$ blackbox_decode --merge-gps LOG009.TXT
Decoding log 'LOG009.TXT' to 'LOG009.01.csv'...

Log 1 of 1, start 01:55.192, end 03:56.387, duration 02:01.195

Statistics
Looptime            516 avg            7.1 std dev (1.4%)
I frames    7331  125.1 bytes avg   917282 bytes total
P frames   21987   81.2 bytes avg  1785252 bytes total
H frames      59   10.0 bytes avg      590 bytes total
G frames     672   25.4 bytes avg    17064 bytes total
E frames       1    6.0 bytes avg        6 bytes total
S frames     675   41.0 bytes avg    27675 bytes total
Frames     29318   92.2 bytes avg  2702534 bytes total
Data rate  241Hz  22712 bytes/s     227200 baud

5 frames failed to decode, rendering 3 loop iterations unreadable. 5 iterations are missing in total (2ms, 0.00%)
205254 loop iterations weren't logged because of your blackbox_rate settings (106045ms, 87.50%)
```
and (decode consumer)
 
```
$ flightlog2kml LOG009.TXT 
Firmware : INAV 3.1.0 (cfcf39a4) MAMBAF722_I2C of Jul 21 2021 12:03:13
Size     : 2.62 MB
Log      : LOG009.TXT / 1
Flight   : Diatone Roma L5 on 2021-08-13 20:58:02
Altitude : 69.8 m at 01:07
Speed    : 22.0 m/s at 01:06
Range    : 179 m at 01:10
Current  : 67.7 A at 01:06
Distance : 996 m
Duration : 02:01
Disarm   : Switch
```
Number of CSV lines is:

```
$wc -l LOG009.01.csv 
29869 LOG009.01.csv
```

Using current release
```
$ blackbox_decode --help
Blackbox flight log decoder by Nicholas Sherlock (v0.4.6, Jun 10 2021 22:05:55)
```
(Arch Linux x86_64 fwiw). 

Please confirm you're using the latest blackbox-decode release and your operating system.

stronnag, 2021-08-15:
KMZ visualisation from flightlog2kml / blackbox_decode 0.4.6.
[image]

Looks sane to me.

avsaase, 2021-08-15:
I was using 0.4.5, updating to 0.4.6 solved the issue. Not sure why this is a problem with this log and not with another one I recorded a few days earlier but in either case it is solved. Thanks.

closed 2021-08-15 by avsaase

stronnag, 2021-08-15:
Very strange, 0.4.5 seems OK here. Howvever, 0.4.6 will be safer as it understands the current f/w modes and adjustments, the latter at least could cause earlier blackbox_decode versions to fail.
```
$ ~/bin/blackbox_decode-0.4.5 --merge-gps LOG009.TXT 
Decoding log 'LOG009.TXT' to 'LOG009.01.csv'...

Log 1 of 1, start 01:55.192, end 03:56.387, duration 02:01.195

Statistics
Looptime            516 avg            7.1 std dev (1.4%)
I frames    7331  125.1 bytes avg   917282 bytes total
P frames   21987   81.2 bytes avg  1785252 bytes total
H frames      59   10.0 bytes avg      590 bytes total
G frames     672   25.4 bytes avg    17064 bytes total
E frames       1    6.0 bytes avg        6 bytes total
S frames     675   41.0 bytes avg    27675 bytes total
Frames     29318   92.2 bytes avg  2702534 bytes total
Data rate  241Hz  22712 bytes/s     227200 baud

5 frames failed to decode, rendering 3 loop iterations unreadable. 5 iterations are missing in total (2ms, 0.00%)
205254 loop iterations weren't logged because of your blackbox_rate settings (106045ms, 87.50%)

$ wc -l LOG009.01.csv
29869 LOG009.01.csv

```