iNavFlight/blackbox-tools #58: Wrong value given by  INAV Blackbox tools for 5.0.0

Torchi, 2022-10-08:
I get a wrong value (-2147483648) of "energyCumulative (mAh)" in the second flight of a blackbox  file produced by INAV 5.1.0.

Attached the blackbox.
[blackbox_log_2022-10-08_114352.TXT](https://github.com/iNavFlight/blackbox-tools/files/9739678/blackbox_log_2022-10-08_114352.TXT)

stronnag, 2022-10-08:
Looks like the FC reports -ve usage, and the decoder sign extends it incorrectly. 
Thanks for the log, I'll take a look.

stronnag, 2022-10-08:
Can you also please supply a log with this problem using the official 5.1.0 build?

Torchi, 2022-10-08:
Yes.  I can produce a log of a similar flight with the official build once I go to fly next week  (note:  I had to  rebuild the FW  but the changed code is very small, just adding the possibility to use full power of the AKK VTX)

stronnag, 2022-10-08:
OK, but it's probably a firmware bug; the BBL just reports what the FC provides.

stronnag, 2022-10-08:
>  OK, but it's probably a firmware bug; the BBL just reports what the FC provides.

More likely it is decoder bug, in the way it integrates current over time.

PR #59 (2022-10-08): initialise energy accumulator for each log segment (#58)
Fix #58 by initialising the current accumulator for each log segment
closed 2022-10-08 by stronnag