iNavFlight/blackbox-tools #69: Datetime wrong on gpx generated file

jmcdrone, 2024-01-09:
blackbox_decode genarate a gpx file with not corresponds to the correct datestamp from log.txt file. It always some minutes above the correct time. Like 10, 8 or 3 minutes above correct time. Another observation: The osd displays the correct datetime!!!! :(

stronnag, 2024-01-09:
It's worse that that (misinterpreting the time offset); it also ignores the time zone and pretends everything is UTC.

closed 2024-01-10 by stronnag

jmcdrone, 2024-01-14:
//------------------------gpx file datetime fixed!----------------------------

char *format_gps_timez(flightLog_t *log, int64_t microseconds, char *tbuf, size_t tbufsz) {
    char strTime[100];
    sprintf(strTime, "%ld", log->sysConfig.logStartTime.tv_sec*1000000 + (microseconds-log->firsttime)); 
    long dTime;
    sscanf(strTime, "%10ld", &dTime);
    time_t unix_timestamp = dTime; 
    struct tm *tm_info;
    tm_info = gmtime(&unix_timestamp);
    strftime(tbuf, tbufsz, "%Y-%m-%d %H:%M:%S", tm_info);

    return tbuf;
//----------------------------------------------------------------------

stronnag, 2024-01-14:
It is already fixed in #71 without using strings.

jmcdrone, 2024-01-14:
Now you get the idea! ;)

stronnag, 2024-01-14:
I'm not sure what idea you are taking about, or the need to work with 16 digits. Using strings a waste of time, inefficient and leads to all sorts of bugs. 

• #71 works correctly across all targets and architectures (unlike your  proposal)
• #71 is efficient (no unnecessary string functions)
• Your proposal fails to compile on Windows, MacOS and any 32bit system (`-Wall   -Werror`)
• Your proposal (compiled without `-Werror` on ia32) gives grossly incorrect   results due to naive integer overflow.
• Your proposal fails to generate a valid `xsd:datetime` with decimal seconds   and time zone.

But thanks for raising the initial issue and testing.

PR #70 (2024-01-10): Fix date time GPS offsets
* Fix --datetime GPS offset (startup / arming deltas)
* Improve (POSIX at least) TZ  usage
* Fixes #69