iNavFlight/blackbox-tools #78: Export 3d speed in mph and other units

redjetvark, 2024-03-20:
Current Behavior
Ground speed in various units, alt in cm

Desired Behavior
3D Speed is several units

Suggested Solution
Coded into inav BB decoder

Who does this impact? Who is this for?
All

Please.

stronnag, 2024-03-21:
The purpose of the application is to decode data from the log.

The user can generate derived data from the log as they see fit.

Doing anything else will result in a high maintenance burden as different users have different requirements for derived data.

This can be done by the user either as a spreadsheet calculated column or as a filter that adds an extra column.

stronnag, 2024-03-21:
In any case,  this is irrelevant; the vertical speed already occurs in the log in two places, `navVel2[2]` and `GPS_velned[2]`.

redjetvark, 2024-03-23:
Thanks. Are the units for these fixed or can be set?

closed 2024-11-13 by stronnag