iNavFlight/blackbox-tools #79: Feature request: select mode for stick overlay

RobertoD91, 2024-03-25:
It would be useful to have the option to select a different mode, apart from Mode 2, for the on-screen overlay of the sticks. In Italy, Mode 1 is widely used for model aircraft.

stronnag, 2024-03-30:
@RobertoD91 
There is a PR https://github.com/iNavFlight/blackbox-tools/pull/80 that adds `--mode MODE` option and should draw the sticks according to requested mode (the default mode remains mode 2 for backwards compatibility).

Can you please test this in mode 1. 
Let me know the preferred OS / Architecture if you'd like me to build a binary for you.

RobertoD91, 2024-03-30:
Thank you for the fast implementation! After a quick test (build commit 0fdadf2b4a5f4104f451bd5627a9c047127f9fbb on intel macos 14.4.1) seems perfect!

stronnag, 2024-03-30:
Addressed in #80

closed 2024-03-30 by stronnag