iNavFlight/blackbox-tools #87: Increase FLIGHT_LOG_MAX_SERVOS to 18?

bartslinger, 2024-07-13:
MAX_SUPPORTED_SERVOS in INAV was increased to 18 in this commit https://github.com/iNavFlight/inav/commit/b6f29761270b261290d549bcf02a64ff43d33d61

That breaks the parser, because it assumes 16:
https://github.com/iNavFlight/blackbox-tools/blob/6eb2fe722c93120688eaff5acd3c00956b73b028/src/parser.h#L20

Trivial solution would be to increase FLIGHT_LOG_MAX_SERVOS to 18, but than it can no longer parse old logs. So I'm not sure what would be the best approach here.

stronnag, 2024-07-13:
@bartslinger 

Thanks. I'll fix it, we have much ugliness already and a means to cope.

stronnag, 2024-07-13:
Actually it doesn't break old logs, the `FLIGHT_LOG_MAX_SERVOS` macro means "up to". So the trivial fix is the fix.

Committed to the (now misnamed) `inav8_rcmodes` branch.

Thanks for finding this.

bartslinger, 2024-07-14:
`MAX_SERVOS` also need to be increased:

https://github.com/iNavFlight/blackbox-tools/blob/a957e6e8da27379a094a5214f939254575f4166c/src/blackbox_render.c#L43

Or perhaps there should only be one definition?

stronnag, 2024-07-14:
Done, uses `parser.h` valus

closed 2024-07-14 by bartslinger

mmosca, 2024-07-15:
Does this mean that file based blackbox on  sitl works now?