iNavFlight/blackbox-tools #88: problems getting datetime of records

gcerri89, 2024-07-28:
Hello,

I'm a naive user of INAV, so I'm sorry if my question sounds trivial but I could not find any solution to my problem anywhere. I would like to retrieve the information collected by the logger, especially the datetime of each record. I was able to convert the .TXT file downloaded using the blackbox deconding system on my computer. I obtained four different files after decoding: one with GPS locations, one with accelerometer/magnetometer/etc. data, one with gpx track and a last one called event which I don't understand. If I understood correctly, the time unit reported in the logged data is in microseconds, so I tried to obtain the real datetime using R CRAN using the following commmand: 

as.POSIXct(time_us * 1e-06, origin = "2023-08-05 09:05:02.882", tz = "CET")
transforming the time in seconds and then obtaining the datetime using as origin the datetime reported in the TXT file as Log start datetime. 

Nevertheless, the datetimes do not correspond to what I observe when opening the log in the program INAV blackbox explore. I don't know if the datetime I used as origin is wrong and eventually where I can find logged the correct information. Did someone have the same issue and know how to solve it?

Thank you for any suggestion!

Giulia

stronnag, 2024-07-28:
You should try running `blackbox_decode` with `--datetime` to add an UTC column.
If you still  think  that the information is incorrect, please supply the log for inspection.

gcerri89, 2024-07-28:
Thank you for the help! I tried, but I don't have the option --datetime. When I checked all the possible commands these is what 
it returned: 

Blackbox flight log decoder by Nicholas Sherlock (Jul 28 2024 10:36:43)

Usage:
     ./blackbox-tools/obj/blackbox_decode [options] <input logs>

Options:
   --help                   This page
   --index <num>            Choose the log from the file that should be decoded (or omit to decode all)
   --limits                 Print the limits and range of each field
   --stdout                 Write log to stdout instead of to a file
   --unit-amperage <unit>   Current meter unit (raw|mA|A), default is A (amps)
   --unit-flags <unit>      State flags unit (raw|flags), default is flags
   --unit-frame-time <unit> Frame timestamp unit (us|s), default is us (microseconds)
   --unit-height <unit>     Height unit (m|cm|ft), default is cm (centimeters)
   --unit-rotation <unit>   Rate of rotation unit (raw|deg/s|rad/s), default is raw
   --unit-acceleration <u>  Acceleration unit (raw|g|m/s2), default is raw
   --unit-gps-speed <unit>  GPS speed unit (mps|kph|mph), default is mps (meters per second)
   --unit-vbat <unit>       Vbat unit (raw|mV|V), default is V (volts)
   --merge-gps              Merge GPS data into the main CSV log file instead of writing it separately
   --simulate-current-meter Simulate a virtual current meter using throttle data
   --sim-current-meter-scale   Override the FC's settings for the current meter simulation
   --sim-current-meter-offset  Override the FC's settings for the current meter simulation
   --simulate-imu           Compute tilt/roll/heading fields from gyro/accel/mag data
   --imu-ignore-mag         Ignore magnetometer data when computing heading
   --declination <val>      Set magnetic declination in degrees.minutes format (e.g. -12.58 for New York)
   --declination-dec <val>  Set magnetic declination in decimal degrees (e.g. -12.97 for New York)
   --debug                  Show extra debugging information
   --raw                    Don't apply predictions to fields (show raw field deltas)

Maybe I have a wrong version of the blackbox decoder?

stronnag, 2024-07-28:
If you have an ancient version of `blackbox_decode` you should not be surprised if it does not work with contemporaneous firmware.

stronnag, 2024-07-28:
If you're using the Betaflight version you should be even less surprised if it does not work with contemporaneous INAV firmware.

closed 2024-07-28 by stronnag